flathub/org.openscad.OpenSCAD #3: Update runtime to 5.12

LeandroStanger, 2019-03-01:


goncalossilva, 2019-07-23:
I've played with this, but CGAL fails to build due to failing to find MPFR. This works fine in 5.11.

```
== Detect external libraries ==
-- External libraries supported: GMP;GMPXX;MPFR;ZLIB;OpenGL;LEDA;MPFI;RS;RS3;OpenNL;Eigen3;BLAS;LAPACK;QGLViewer;ESBTL;Coin3D;NTL;IPE
-- Preconfiguring library: GMP ...
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmp.so  
-- GMP has been preconfigured:
--   UseGMP-file:      
--   GMP include:      /usr/include
--   GMP libraries:    /usr/lib/x86_64-linux-gnu/libgmp.so
--   GMP definitions:  
-- USING GMP_VERSION = '6.1.2'
-- Preconfiguring library: MPFR ...
-- Could NOT find MPFR (missing: MPFR_LIBRARIES MPFR_INCLUDE_DIR) 
CMake Error at cmake/modules/CGAL_SetupDependencies.cmake:66 (message):
  CGAL requires MPFR to be found
Call Stack (most recent call first):
  CMakeLists.txt:671 (include)
```

t-paul, 2019-12-27:
Closed via #6 (updating to 5.13).

PR #6 (2019-12-29): Update OpenSCAD to 2019.05 (based on previous PR by @goncalossilva)
  * Additionally updates runtime and dependencies
  * Fixes appdata problem

Closes: #3
Closes: #4
Closes: #5
closed 2019-12-27 by t-paul