flathub/org.openscad.OpenSCAD #14: Support for Space Navigator

hammerandtongs, 2021-07-07:
I believe OpenSCAD is affected by the lack of support for the spnav socket in flatpak like Blender and FreeCAD.

https://github.com/flatpak/flatpak/issues/3541

t-paul, 2021-07-07:
The joystick driver seems to work fine, wenn allowing access to the joystick device:
```
flatpak run --device=all org.openscad.OpenSCAD
```

hfiguiere, 2021-07-08:
the spnav is not accessed through a standard device. It's accessed through a socket.

t-paul, 2021-07-08:
Without `spacenavd` installed, the Space Mouse is detected and configured as multi-axis joystick available through `/dev/input/js0` which is supported by OpenSCAD. I'm sure, I wrote the code and I just tested with my Space Mouse Wireless. This works with the command above.

hfiguiere, 2021-07-08:
Adding `--device=all` would acceptable for this situation.

t-paul, 2023-04-18:
Beta version supports spacenavd now. See main [README.md](https://github.com/flathub/org.openscad.OpenSCAD) for installation instructions.

closed 2023-04-18 by t-paul