flathub/org.openscad.OpenSCAD #18: Can we have development snapshots?

Knochi, 2023-02-19:
since openscad stable releases are generated only very rarely (once a year?). It would be very convenient to have the choice to install development snapshots.

t-paul, 2023-02-19:
If someone can point me to documentation on how to do that, maybe. My reading is that flathub does not support this.

So far I failed to find a way to do that in a convenient. I was hoping [OBS](https://openbuildservice.org/2021/02/18/introducing-flatpak-builds/) can jump in here but the process did have too many manual steps for dev snapshots I'm willing to maintain.

Knochi, 2023-02-19:
From my understanding that would be like another app.. e.g. org.snapshot.openSCAD

t-paul, 2023-02-19:
Yes, there's a technical solution for flathub, but it's specifically prohibited by the [Application Requirements](https://github.com/flathub/flathub/wiki/App-Requirements).

> Flathub only hosts stable application releases, and not development snapshots. Manifests should therefore refer to tarballs or git tags and not the tip of a branch.

hfiguiere, 2023-02-19:
Flathub is not for "nightlies" or other dev snapshot.

There is a beta branch for beta releases but its use is a but cumbersome.

If upstream want to build nightly flatpak then there is a github action that allow doing that.

OBS doesn't do snapashot, they publish releases directly.

As for `org.snapshot.openSCAD` it's a no. The domain is completely unrelated to the project.

t-paul, 2023-02-19:
@hfiguiere Thanks, I'll have a look at the github action once time permits...

Knochi, 2023-04-15:
Ok there is this beta flathub

https://discourse.flathub.org/t/how-to-use-flathub-beta/2111

And other project like [freeCad](https://github.com/flathub/org.freecadweb.FreeCAD) use it to publish development snapshots. There is a beta branch in the repository which points to the development release.

t-paul, 2023-04-17:
@Knochi thanks for that info.

A Beta build is now available, although I'm not sure why it does not show up in search results. Maybe it's an indexing thing as it's a totaly new app in a sense.

The buid can be installed directly via:
flatpak install --user https://dl.flathub.org/build-repo/18325/org.openscad.OpenSCAD.flatpakref

hfiguiere, 2023-04-17:
flathub-beta packages are not shown on the website. A no, it's not a new app.

t-paul, 2023-04-17:
I can't find it via command line search, and also the install via flathub-beta says "not found".
```
flatpak search --columns=name,version,remotes OpenSCAD
Name         Version     Remotes
OpenSCAD     2021.01     flathub
```
Seems to work for other apps though
```
flatpak search --columns=name,version,remotes Godot
Name          Version      Remotes
Godot         3.4.4        flathub-beta
Godot 3       3.5.2        flathub
Godot         4.0.2        flathub
godotGem      1.2.0        flathub
```
I'm not seeing anything obviously wrong in the build properties of the last build.
https://buildbot.flathub.org/#/builders/6/builds/35683

hfiguiere, 2023-04-17:
the new beta build hasn't been published yet. It takes 3+ hours

t-paul, 2023-04-17:
Aha! Cool, thank you.

Knochi, 2023-04-18:
Awesome. Will try on my own

closed 2023-04-18 by t-paul