flathub/org.openscad.OpenSCAD #34: [beta] Set center of rotation has an Offset

Knochi, 2024-03-01:
## Problem Description
When i do a double click on some feature the crosshair is placed with an offset. 
Actually I think the doubleclick is registered with an offset.

### Pictures
[image]
I doubleClicked at [50,50] , the Offset is approximally [-16,+15], when i zoom out it's more, when i zoom in it's less. So i guess it'S an offset in screen pixels.

[image]
When i doubleClick into an empty space sometimes it sets the crosshair with an offset. That is why i think the doubleClick is already registered with an offset.

## My Environment
### OpenSCAD
OpenSCAD Version: 2023.08.29
System information: Linux 6.1.52-valve9-1-neptune-61 #1 SMP PREEMPT_DYNAMIC Wed, 22 Nov 2023 23:43:34 +0000 x86_64 KDE Flatpak runtime 8 CPUs 14.47 GB RAM
User Agent: OpenSCAD/2023.08.29 (Linux x86_64; KDE Flatpak runtime)
Compiler: GCC "12.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_82
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.10
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.72.4
lodepng version: 20210627
libzip version: 1.10.0
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 5.2.0
cairo version: 1.16.0
lib3mf version: 2.2.0

### System (SteamDeck LCD 256GB)
Operating System: SteamOS 3.5.7
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.107.0
Qt Version: 5.15.9
Kernel Version: 6.1.52-valve9-1-neptune-61 (64-bit)
Graphics Platform: X11
Processors: 8 × AMD Custom APU 0405
Memory: 14,5 GiB of RAM
Graphics Processor: AMD Custom GPU 0405
Manufacturer: Valve
Product Name: Jupiter
System Version: 1

hadess, 2024-07-29:
This isn't the repository for OpenSCAD, it's the repository for the Flatpak packaging of OpenSCAD. This is the wrong place to report what looks to be a bug in the software itself, not something related to the packaging.

You want to use this repository: https://github.com/openscad/openscad

closed 2024-12-30 by t-paul

Knochi, 2024-12-30:
It's fixed somehow. Don't know on what end, maybe SteamOS maybe openSCAD, but works now.

t-paul, 2024-12-30:
Ah, I closed due to the upstream issue, but that's even better, so we can close that one too?

There was a minor Qt update, maybe that's responsible. Otherwise there's no change on the OpenSCAD side that seems related.